TeamPracticalProjects/LoRa_experiments #39: ENH: Add header to allow programming ATmega

jschrempp, 2024-12-02:
There are 4 lines between the RPi programmer and the ATmega. If we had a header for those 4 on the PCB, then we could make a cable and program the ATmega while it is in place on the PCB.

BobGlicksman, 2024-12-03:
We already have the pins brought out to header fields on the current PCB.  Unfortunately, the UNO is 5 volts and we are limited to 3.3 volts on the development board because of the LoRa module.  Therefore, we cannot hook up an Uno to our PCB.  We will have to continue to remove the ATmega chip from the 3 volt PCB and place it into a 5 volt programmer setup in order for this to work.

jschrempp, 2024-12-04:
I wonder if we could jumper out the LoRa module power so that we could program on the board. Maybe a jumper that in one position powers the LoRa module, but in the other position jumpers offboard power to the rest of the hardware.

BobGlicksman, 2024-12-04:
I'm sure that we could do something along these lines, but why?  When we wanted to program the microcontroller we would need to remove this jumper, and perhaps one on the reset line as well, and of  course disconnect the 3 volt battery,  then connect up 6 wires from the Arduino, and after programming, we would need to remember to remove the wires from the Arduino (or at least the Arduino power) and replace the jumper(s) - and in order, or else we would short the +5 volt supply to the battery and/or connect the LoRa module to +5 and possibly damage it, etc. etc.  And, all of this to avoid popping the microcontroller out of one zif socket and into another for programming?  I'm not sure I see any simplification here, and certainly I see less safety.

closed 2024-12-24 by BobGlicksman